kontakt/Octoprint-Filament-Reloaded #1: Installing this seems to have broken octoprint's settings window

StevenCregan, 2017-01-13:
After installing, I noticed that octoprint's settings modal form seems to be broken.
Everything is now displaying as a single page instead of tabs
[image]

StevenCregan, 2017-01-13:
Think I may have done this myself while figuring out why the plugin wouldn't install properly for me

StevenCregan, 2017-01-13:
Yes, this was my mistake. Trying to install via the plugin manager was failing, so I tried adding the --upgrade modifier to see if that would fix it. It did not, and in addition, it broke Octoprints formatting due to being on a different version of Jinja.

The error I was having is that RPi.GPIO wasn't installed. Please consider adding that as a dependency for PIP

closed 2017-01-13 by StevenCregan

kontakt, 2017-01-13:
I've added that to the package. Thanks for the heads up! I also managed to mess up my own formatting with the same --upgrade flag. How did you revert jinja?

StevenCregan, 2017-01-13:
I ssh'd into the pi, viewed what packages were changed in `.octoprint/logs/plugin_pluginmanager_console.log`
then ran
`/home/pi/oprint/bin/python -m pip install Jinja2==2.8`
as well as reverting all the others changed, I could tell which ones got changed by finding the line that said
>Installing collected packages: Octoprint-FilamentReload, Flask-Login, netaddr, netifaces, rsa, psutil, Jinja2, certifi, pytz, Babel, webassets, argh, regex, Unidecode