kontakt/Octoprint-Filament-Reloaded #3: Support for dual-extrusion

zorrobyte, 2017-03-13:
It would be nice to support two filament sensors for dual extrusion setups. I may submit a PR sometime in the future if I get the free time.

kbastronomics, 2017-04-06:
as a work around you can wire your two sensors in series using normally closed switches.  if either switch trips it will trigger the pin and printing would halt.

kontakt, 2017-04-06:
That's a decent tip for simple setup. I hope to try and hook into the active extruder so you don't have to have both loaded to print anything.

mark-hahn, 2017-07-26:
> so you don't have to have both loaded to print anything.

You only need to stick a couple of inches or so of filament in one of the sensors and tie it in a knot.  Not a big deal.

kontakt, 2017-08-02:
It's not a huge deal, but it is one of those quality of life things that make something annoying to use.

kbastronomics, 2017-08-03:
i think the biggest issue in this type of support is complexity.   with a simple series switch arrangement you can not monitor them independently which you would need to do if you implemented true dual (or more) filament monitors.   so you get into either use 2 or more inputs which the Pi has but now you must track how many extruder's are active, which extruder is active.  if more than one is active but not in use do you disable the sensor.  the number of possible situations to code for goes up.

the simple series switches and as was said. put a small piece of filament in one that's not in use is the simple solution that give the user the most control.   coding for 2 or more is doable.  but is it worth it at this time.   

just my opinion on the subject not to be confused with fact :D

cpmorris, 2017-10-16:
Just to be clear, for dual sensing I need two external relays to combine the signal?  NO way to do it otherwise?  THanks

rpmoles, 2017-10-30:
I have been working with the first example of placing the two filamen sensors in series. The only way I can see is an example on the PI using pin 18-GPIO 24 and 20 ground. Worked ok with one sensor. But when I tried using two the only way it worked if I removed both filaments to kick off a signal.  The sensors I am using are https://www.amazon.com/gp/product/B0748C4CS5/ref=oh_aui_detailpage_o04_s00?ie=UTF8&psc=1.
I connected them by pin 18 switch one white, switch one black to switch two white, and switch two black to pin 20 ground. The second way with relays really like to know how to do that. I have a bank of four, and only using one for controlling dual fans for cooling prints.

rpmoles, 2017-10-30:
You know I will try something else with my sensors. I think the saying "can't see the forest through the trees" fits. After I looked at what I wrote down and sent, I about kick myself. I can see with the settings in octoprint. I should have used the red and black wires on the sensors. With using the white and black I can see why I was getting the results I was getting.  Will try again later today.

rpmoles, 2017-10-30:
I tried various ways with just the two sensors. I believe part of the issue is power draw.   When I used only one sensor and when there is filament inserted the sensor lights up bright. When there are two in series there ls no lights. However when I connect each one to different 3.3v  pin 1 and 17 both are lights are bright.

kbastronomics, 2017-10-31:
looking at that sensor is it's not a simple switch.  its acting like they
did a simple logic level output and not a open collector one so 2 won't
work in series if I am correct.
without seeing the actual circuit though I can not say for sure

rpmoles, 2017-10-31:
BIQU did something with them. Well either tear em open, get different switches. or go with a previous idea and hook up relays. I will go with the relay route.

rpmoles, 2017-11-02:
[image]
I have the dual filament sensors working.  The relays worked good and took a bit to get the logic right.
So if one or the other runs out it will pause the print.
[Filament-Sensors.pdf](https://github.com/kontakt/Octoprint-Filament-Reloaded/files/1436183/Filament-Sensors.pdf)

rpmoles, 2017-11-02:
On the sketch forgot to add the jumper across relay 1 common to relay 2 NC

KC703, 2018-01-13:
Interested in the relay mod.  What type of relays were used?

Thanks!

rpmoles, 2018-01-13:
The relays I purchased off of Amazon. **SunFounder 4 Channel 5V Relay Shield Module for Arduino UNO 2560 1280 ARM PIC AVR STM32**.  I had other functions on top of dual filament sensors.
Yes the jumper is installed that what was needed for control power for the relay.
I am rescanning the updating docs and will add shortly.

rpmoles, 2018-01-13:
I have also researched gcodes and have this added to the octoprint ggode scirpts.
after print is paused
M117 Print Paused

{% if pause_position.x is not none %}
; relative XYZE
G91
M83

; retract filament, move Z slightly upwards
G1 Z+5 E-5 F4500 

; absolute XYZE
M82
G90

; move to a safe rest position, adjust as necessary
G1 X0 Y100
{% endif %}

before print job is resumed
M117 Print Resumed

{% if pause_position.x is not none %}
; relative extruder
M83

; prime nozzle
G1 E-5 F4500
G1 E5 F4500
G1 E5 F4500

; absolute E
M82

; absolute XYZ
G90

; reset E
G92 E{{ pause_position.e }}

; move back to pause position XYZ
G1 X{{ pause_position.x }} Y{{ pause_position.y }} Z{{ pause_position.z }} F4500

; reset to feed rate before pause if available
{% if pause_position.f is not none %}G1 F{{ pause_position.f }}{% endif %}
{% endif %}

rpmoles, 2018-01-13:
It worked even with different printers. Had my mod davinci. Now I also have a  Tronxy P802MA that works with it.

Here is the updated schematic 1 is update the pdf you were looking at, and 2 is the update with all my 
[schematic-2.pdf](https://github.com/kontakt/Octoprint-Filament-Reloaded/files/1628442/schematic-2.pdf)
[schamtic-1.pdf](https://github.com/kontakt/Octoprint-Filament-Reloaded/files/1628443/schamtic-1.pdf)

gadgets added temp/humidity sensor, etc...

KC703, 2018-01-14:
Thanks for the confirmation!

Totoleheros, 2019-05-04:
@rpmoles: for those, like me, that are not experts in electronics, could you please post a photo of the final wiring on the relay ? Thank you in advance!!

Totoleheros, 2019-05-21:
@rpmoles : Forget my request, I could make it work with your indications. Thank you for sharing your solution.

joaomamede, 2019-07-21:
Hi, the rpi (from v2 onwards) has two gpio pins. When I have a bit of time, I guess I'll try to make a redo of this plug-in to make it work without having to use other electronics.