kontakt/Octoprint-Filament-Reloaded #8: Refresh Pin, Debounce and Open/Close when updated

kbastronomics, 2017-04-06:
it looks like you only check the pin status on start up so that if after starting with the default Pin of -1 then changing the Pin to a valid value the plugin is not refreshed.

I was trying to figure out why it's not working and it looks like its because of this reason
I don't see the pin status change.  I am using GPIO18 Pin 12 to ground thru a NO switch.  so it should be working

can you look at some kind of setting refresh so changes are more dynamic

kontakt, 2017-04-06:
Having the settings require a restart is a bit much. I'll fix this.

ChrisMayhew, 2017-05-05:
I'm having issues with this Plugin not working, set the pin and just doesn't respond.. The sensor used to work with the original filament sensor plugin.
Any updates due?

kbastronomics, 2017-05-06:
Which pin numbering did you use.  the original used BCM (GPIO) and this now uses Board (PHYSICAL) which mean if you used for example GPIO 18 before it is now PIN 12.
[GPIO/PIN OUT](http://www.raspberrypi-spy.co.uk/wp-content/uploads/2012/06/Raspberry-Pi-GPIO-Layout-Model-B-Plus-rotated-2700x900.png)

ChrisMayhew, 2017-05-06:
Ah that's explains it, I used 23 so I guess that's 16 physical? I'll have a go when I get home from work!

kbastronomics, 2017-05-06:
Yes, GPIO23 would be Pin 16

ChrisMayhew, 2017-05-06:
Works on Pin 16 now after restarting OctoPrint but it keeps pausing the print all the time.. I set the bounce to 10000 and it's still doing it, Do I have to restart every time I adjust any settings?

kbastronomics, 2017-05-06:
Yes 
Settings are currently only read at startup

ChrisMayhew, 2017-05-10:
No matter what I set on the settings I can't get this to work properly, Straight away as soon as I start a print it pauses and just constantly pauses. I have the bounce set at 10,000 and still does it.

kbastronomics, 2017-05-10:
for debounce anything past the default really won't help much so that does
not sound like your problem.  youd have to a switch that is really ringing
and never stops bouncing for the value you already have.

what do you have switch type set to?   have you tried changing that.  it's
either Normally Open or Normally Closed depending on the switch you used.

your switch should connect from GND thru your switch to the PIN your using.
if you have it going to 5V instead and with the pullups used on the input
pin you'll never see the switch change state.

Do yo have a multimeter you can check it with. a cheapy will do for this.
you just need to see the pin go HIGH/LOW depending on the filament.

finally remember to reboot Octoprint anytime you change any setting related
to the filament plugin

ChrisMayhew, 2017-05-10:
I had it set to Normally Open, but I'm sure i tried Normally Closed too.

My switch is one of them optical end stops and has three wires, one got GND, Signal and 5v line?

I'll check it with a multimeter tomorrow.

kbastronomics, 2017-05-10:
oh i see.
those operate a bit differently.   do happen to have the model number of
the optical switch.  I can look it up and see if it would work correctly.

ChrisMayhew, 2017-05-10:
I got it from this eBay page http://www.ebay.co.uk/itm/161780649665

Not sure if that will help any?

ChrisMayhew, 2017-05-10:
I've just seen a YouTube video of someone who looks as if they're using the same sensor and using this plugin too, and they had it set to Normally Closed, I'll try that again tomorrow after this print finishes and report back.

ChrisMayhew, 2017-05-11:
It's definitely better, but I've still had 2 pauses in 20 minutes of printing.

kontakt, 2017-05-11:
So you're using an optical endstop with the filament feeding through the middle? What's your alignment like? If it drifts too far off center you will get an open signal.

ChrisMayhew, 2017-05-12:
Yes, It's running through a case for the end stop at the moment, I'll check how much movement it has and maybe print a different case if I need too

kontakt, 2017-05-13:
I would definitely put a multimeter or even better an oscilloscope on your filament sensor output and give it a good wiggle to see what sort of signal you see.

ChrisMayhew, 2017-05-13:
Having modded the guys case design to make the hole smaller it seems to be working ok now! I made it so it's just big enough for 1.75 give or take 0.10 or so for the tolerance of the filament and it printed a full 4 hour print without a pause! 👌👍🏻

kontakt, 2017-05-13:
Awesome.

Hypertoken, 2017-07-25:
@ChrisMayhew Happen to have that modded case? I'm having the same issue > sensor constantly switching open/closed

2017-07-24 13:49:36,201 - octoprint.plugins.filamentreload - INFO - Out of filament!
2017-07-24 13:49:36,203 - octoprint.plugins.filamentreload - INFO - Pausing print.
2017-07-24 13:49:36,818 - octoprint.plugins.filamentreload - INFO - Filament detected!
2017-07-24 13:49:37,302 - octoprint.plugins.filamentreload - INFO - Out of filament!
2017-07-24 13:49:37,304 - octoprint.plugins.filamentreload - INFO - Pausing print.
2017-07-24 13:49:37,742 - octoprint.plugins.filamentreload - INFO - Out of filament!
2017-07-24 13:49:37,744 - octoprint.plugins.filamentreload - INFO - Pausing print.
2017-07-24 13:49:38,145 - octoprint.plugins.filamentreload - INFO - Filament detected!
2017-07-24 13:49:38,545 - octoprint.plugins.filamentreload - INFO - Out of filament!
2017-07-24 13:49:38,552 - octoprint.plugins.filamentreload - INFO - Pausing print.
2017-07-24 13:49:38,946 - octoprint.plugins.filamentreload - INFO - Filament detected!

ChrisMayhew, 2017-07-25:
@Hypertoken I use this one now https://www.thingiverse.com/thing:2300359
I combined two files to allow it to fit my I3 Plus easily so it might not work for you without modding it.