kontakt/Octoprint-Filament-Reloaded #10: Getting the sensor to work

MPTool, 2017-05-31:
I have set up my filament sensor to use pin #12 and for some reason when I try to test it by pulling the filament through the led light lights up but it does not want to stop the print.  I have updated my Raspberry PI and gone through every step that you have in the description including trying to set it up for open or closed and neither work.  https://www.amazon.com/gp/product/B00EVOKHPO/ref=oh_aui_detailpage_o07_s00?ie=UTF8&psc=1  that is the sensor that I am trying to use and https://www.thingiverse.com/thing:1789486  is the case that i am using for this sensor but it seems like no matter what I do the sensor and your code wont stop the print.  Thank you so much for taking the time to read this and I hope to hear from you soon.

kontakt, 2017-06-12:
Sorry for the late reply, I have been out of town. Are you still having this issue?

MPTool, 2017-06-12:
No problem and sadly yes I am.  I have tried everywhere to try and get these sensors to work but it feels like it is something in the code because when there is filament present in the sensor the light is off then when I take the filament out of the sensor the light turns on.  I am honestly really lost on what in the world it could be.  I'm using the lulzbot taz 6 with the sensor that I attached in the original comment and a raspberry pi b+.  if you need more information on this please let me know.  I'm fairly new to 3D printing and haven't had a lot of luck with asking people who do this all the time.

MPTool, 2017-06-12:
The ground is wired to GROUND Pin 14, the signal is plugged into the GPIO 18 (or pin 12), and the volt is plugged into the volt is plugged into the very first 5v power pin. using the diagram on: http://www.raspberrypi-spy.co.uk/wp-content/uploads/2012/06/Raspberry-Pi-GPIO-Layout-Model-B-Plus-rotated-2700x900.png .

kontakt, 2017-06-12:
Try using pin 12 in the settings, by default the system uses the board pin numbers rather than bcm.

MPTool, 2017-06-12:
I'll give that a try right after this quick calibration print.  Thank you so much for replying again and ill keep you updated on if that helps me.

MPTool, 2017-06-12:
It now pauses the print but the problem now is that it will actually not allow me to print anything at all now.  When I disable the plugin though it allows the print to fully print.

kontakt, 2017-06-14:
Alright, progress. I'm working on a new feature that shows status of the sensor. I'll hopefully have that finished in a week. Do you have a multimeter?

MPTool, 2017-06-14:
Currently I do not because I am setting this all up at work but i can bring one in tomorrow if that can help with testing.

MPTool, 2017-06-15:
Is there a special code that I need to put into the GCODE or something?  because if not I'm really confused with what could be going wrong.

kontakt, 2017-06-16:
I'm not sure either. If it's pausing then it's receiving a signal. I want you to check the output voltage line of the sensor with and without filament.

MPTool, 2017-06-19:
Would it be a bad thing to run test prints without filament in the tool head just so I can see if the sensor is working or not?

kontakt, 2017-06-19:
Sure, if your software supports it, you maybe be able to run prints without extruding anything or enabling your heater. Check your settings.

MPTool, 2017-06-19:
I just wanted to make sure that it wouldn't mess any part of this printer up.  Hopefully I can take apart my sensor and try to run test prints on it without having to waste any materials.

MPTool, 2017-06-19:
It looks like I have finally hit the issue and corrected it.  The problem was that the sensor was over clamped causing the wiring inside to not be in its proper areas which made the sensor able to detect the filament but it was not able to send the signal to the program to pause it.  
Thank you so much for all of your help again.  I just have one little question to ask then you can close this one all up.  Do you know what GCODE I could put in the before print resumes to allow the toolhead to go back to its original position if I am using the M600 code after the print pauses?

kontakt, 2017-06-20:
M600 is supposed to return to the previous print location and resume after you click the display on the printer. I haven't used it before, so I don't know exactly how. 

I'm glad you have the sensor sensing!

closed 2017-06-20 by kontakt