kontakt/Octoprint-Filament-Reloaded #13: M600 Command

ReadieS, 2017-07-03:
Is it possible to use the M600 command with this plugin to run the script built into Marlin. I have set the plugin up on Octoprint but when I test the runout reaction it runs the M600 script perfectly but after returning to the print location it immediately repeats the whole M600 process again and continues to loop making it impossible to resume printing.
Thank you for all your hard work on the plugin, I apologise I have no experience writing code and as such don't understand the implications of what I am asking please feel free to ignore if I am making a stupid request.
Thanks,
ReadieS

kontakt, 2017-07-22:
Sorry for the very slow response; you are triggering the M600 through octoprint's pause script?

Hypertoken, 2017-07-30:
I am having the same issue! I'm currently in process of figuring it out 😛 

If I setup the plugin pause and use M600 it will act in the above manner, repeating over and over, never restarting.

Currently I have removed my M600 script, and I'm only using the pause function, but I don't care for this as the nozzle just sits over the print right where it paused, at full temp.

This is my M600... ( M600 X100 Y15 Z10 E0 L25 ) 
I like M600 because it parks my extruder, and if I'm not right there to refill it, it will turn off the hotend. 

Currently I have to monitor it, and if I see it's paused, then I have to then manually send the M600. Doing it this way by pausing via the plugin, and then manually sending the code works, and will resume properly.

ReadieS, 2017-08-02:
Hi,
Sorry for the long delay in replying its a pretty busy time at the moment for me but thank you for your reply Kontakt and your input Hypertoken. I have done a little playing around with the settings.
a) With M600 set in the plugin and it not set to pause printing results in a loop of M600
b) With M600 Set in the Plugin with it set to pause printing it runs m600 once but doesn't resume printing unless you press resume in octoprint (not ideal but at least workable)
c) With no script in the plugin and M600 in octoprints pause script results in the same as b) it runs once but doesn't resume (even less ideal because its a pain to go right the way through the M600 command if you only wanted to pause your print for a few seconds)
It would be ideal if when M600 was entered in the Gcode Script it could be made to auto resume after one run through as long as the filament sensor showed filament loaded.
I'm afraid I have no coding experience other than editing marlin configs so I have no idea how hard this is to implement. Really appreciate the work you have put into the plugin though, if you need me to check anything my end I would be more than happy to.

kontakt, 2017-08-02:
I need to look into how the gcode is handled by octoprint after we call it, but if I can get a callback this shouldn't be too complex.

totalitarian, 2017-08-31:
Any update on this?

devsfan1830, 2017-10-18:
I got stuck in an M600 loop. Placed the M600 command in the plugin, "pause" checkbox was unchecked.

pablosimone, 2018-01-15:
I have the same problem. Have to disable the plugin because it does not let me print anything.

rmeijsSdu, 2018-02-20:
## Yeah, I got it working on Marlin 1.1.8

###  Octoprint Filament Reloaded plugin setttings:

The M600 command in Marlin sends a pause (and later resume) to Octoprint (see below). Therefore we don't want the plugin to pause.
```
Out of filament GCODE: M600
Pause print when out of filament: NO (unticked)
```

### Settings in Configuration.h (Marlin)

Since we use Octoprint's Filament runout sensor and not Marlin's own:
```
//#define FILAMENT_RUNOUT_SENSOR
```
Needed for ADVANCED_PAUSE_FEATURE (M600)
```
#define NOZZLE_PARK_FEATURE
```

### Settings in Configuration_adv.h (Marlin)

I didn't want my printer to home first before filament change since my z-axis homing might bump into an existing print. You might want it different.
```
#define ADVANCED_PAUSE_FEATURE
#if ENABLED(ADVANCED_PAUSE_FEATURE)
  #define PAUSE_PARK_RETRACT_FEEDRATE 60      // Initial retract feedrate in mm/s
  #define PAUSE_PARK_RETRACT_LENGTH 2         // Initial retract in mm
                                              // It is a short retract used immediately after print interrupt before move to filament exchange position
  #define FILAMENT_CHANGE_UNLOAD_FEEDRATE 10  // Unload filament feedrate in mm/s - filament unloading can be fast
  #define FILAMENT_CHANGE_UNLOAD_LENGTH 100   // Unload filament length from hotend in mm
                                              // Longer length for bowden printers to unload filament from whole bowden tube,
                                              // shorter length for printers without bowden to unload filament from extruder only,
                                              // 0 to disable unloading for manual unloading
  #define FILAMENT_CHANGE_LOAD_FEEDRATE 6     // Load filament feedrate in mm/s - filament loading into the bowden tube can be fast
  #define FILAMENT_CHANGE_LOAD_LENGTH 0       // Load filament length over hotend in mm
                                              // Longer length for bowden printers to fast load filament into whole bowden tube over the hotend,
                                              // Short or zero length for printers without bowden where loading is not used
  #define ADVANCED_PAUSE_EXTRUDE_FEEDRATE 3   // Extrude filament feedrate in mm/s - must be slower than load feedrate
  #define ADVANCED_PAUSE_EXTRUDE_LENGTH 50    // Extrude filament length in mm after filament is loaded over the hotend,
                                              // 0 to disable for manual extrusion
                                              // Filament can be extruded repeatedly from the filament exchange menu to fill the hotend,
                                              // or until outcoming filament color is not clear for filament color change
  #define PAUSE_PARK_NOZZLE_TIMEOUT 45        // Turn off nozzle if user doesn't change filament within this time limit in seconds
  #define FILAMENT_CHANGE_NUMBER_OF_ALERT_BEEPS 5 // Number of alert beeps before printer goes quiet
  #define PAUSE_PARK_NO_STEPPER_TIMEOUT       // Enable to have stepper motors hold position during filament change
                                              // even if it takes longer than DEFAULT_STEPPER_DEACTIVE_TIME.
//  #define PARK_HEAD_ON_PAUSE                // Go to filament change position on pause, return to print position on resume
//  #define HOME_BEFORE_FILAMENT_CHANGE       // Ensure homing has been completed prior to parking for filament change
#endif
```
These are probably the settings you have overlooked:
```
#define ACTION_ON_PAUSE "pause"
```
```
#define ACTION_ON_RESUME "resume"
```