kontakt/Octoprint-Filament-Reloaded #18: Continually Pauses

HarrysWorkshop, 2017-09-27:
Problem.  When set up and run, it will run fine for a minute or two, then it pauses.   Shows it pauses on Octoprint.  I can reset it and it continues, only to pause in a minute or two.  I looked at the G code sent and it only shows that it has been paused, no indication as to why..  
System is A6 with Octopi wireless.  Pin used is 4 with normally open contact.  Connected to ground when there is filament.  I am directly connected with alligator clips, no switch.  It does pause when I unclip clips, and when clipped and hit resume it restarts, only to pause again in a few minutes.  I am running latest version of Skynet with autobed leveling.   Everything works and does not pause when I remove your program.  Keep awake on Skynet has been activated.  I have reloaded Skynet and Octopi twice with same results (pausing).  Any suggestions on what to do?  Do you have a basic program with out all the features that I can test?  You know, Just set the Pin number, bounce and NO or NC  Regards, Harry Whitehead   email is hmwhvs at aol dot com

kontakt, 2017-11-11:
Sounds like better debouncing is needed. I'll see what I can do.

HarrysWorkshop, 2017-11-11:
Let me know if you need more info.  My thoughts are it is somewhere in how the raspberry Pi communicates with the A6, ie  why is the G600? code sent from the raspberry at odd times, when the pin is grounded.  I have not replaced the raspberry yet, but do have one and will try that.  Other wise, raspberry is working fine.  Regards, Harry

HarrysWorkshop, 2017-11-11:
Opps, meant to say when pin is not grounded.

closed 2017-11-11 by HarrysWorkshop

kontakt, 2017-11-11:
Do you have a diagram with your layout? After re-reading that, you might just need a pull-up. If you have a switch that is open-circuit, with a floating GPIO line without filament the input is undefined.

HarrysWorkshop, 2017-11-11:
Yes, I have a pull up resistor on the pin when switch is NO.  I just changed out my raspberry today but have not installed the filament sensor yet.  Will do that tomorrow or Monday and let you know what the outcome is.  Regards,  Harry