kontakt/Octoprint-Filament-Reloaded #22: Ignore sensor when print is paused.

NeoCortex3, 2017-11-24:
Is there a way that the plugin can ignore the sensor when the print is allready paused?

I keep pressing my sensor-button when i put in new filament and that runs my pause-script again even the print is allready paused.

mylife4aiurr, 2018-03-08:
Yes this is super frustrating. After the out of filament is detected, the pluggin will pause:
"Sending out of filament GCODE"
so I physically change the filament......
while I'm changing the filament the pluggin is still logging or saving the switches state changes. So what ends up happening is the printer resumes printing but immediately does another filament change because it was registering/saving the switch state changes during the filament change. It's like its stuck in a loop, because will just keep looping filament changes. I end up just cancelling the print. 
I have tried increasing debounce, but it has no effect. The output below is from the Octoprint logs. Only the 1st out of filament is from actually out of filament, the others were while I was changing filament. 

```
 2018-03-08 04:02:59,708 - octoprint.plugins.filamentreload - INFO - Setting up sensor.
2018-03-08 04:02:59,710 - octoprint.plugins.filamentreload - INFO - Using BCM Mode
2018-03-08 04:02:59,711 - octoprint.plugins.filamentreload - INFO - Filament Sensor active on GPIO Pin [24]
2018-03-08 04:03:27,388 - octoprint.plugins.filamentreload - INFO - PrintStarted: Enabling filament sensor.
2018-03-08 04:03:27,433 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2018-03-08 04:05:28,432 - octoprint.plugins.filamentreload - INFO - Out of filament!
2018-03-08 04:05:28,438 - octoprint.plugins.filamentreload - INFO - Sending out of filament GCODE
2018-03-08 04:05:29,101 - octoprint.plugins.filamentreload - INFO - Out of filament!
2018-03-08 04:05:29,102 - octoprint.plugins.filamentreload - INFO - Sending out of filament GCODE
2018-03-08 04:05:47,202 - octoprint.plugins.filamentreload - INFO - Out of filament!
2018-03-08 04:05:47,204 - octoprint.plugins.filamentreload - INFO - Sending out of filament GCODE
```

mylife4aiurr, 2018-03-08:
Just a follow up from last night. I still have problem of filament change loop.

I used M600 in the filament reloaded pluggin page, with pause print unchecked.

DrWatson, 2018-06-04:
I am wondering about adding something similar as I noticed the loop also. In my case though, I added some gcode that would do G1 Z10 to move the head off the printed part. I wasn't even messing with the switch or attempting a reload and the carriage kept moving up and up and up. 

I suspect it may have to do with some polling the plugin is doing and detecting the the switch is still open and re-executing the Gcode again rather than what you suggest that it's the toggle from open to closed causing the Gcode to execute. Maybe a subtle difference from what you are saying, I think.

I am going to try doing the following instead as I didn't try and pause the print in Gcode. You are using the filament change gcode and I am going to try the pause print command instead. Did your solution above work?

Here is the code I am going to try:
```
M226          ;pause print
M104 S0       ;turn extruder heater off
G91           ;change the printer into relative movement mode
G1 Z19        ;move the head up 10mm
G90           ;put printer back in absolute movement mode

```
And then, of course, unchecking the pause print option in the plugin.

Do you think this will work?

DrWatson, 2018-06-06:
I have made a pull request that contains a fix that addresses this problem I believe. See:

https://github.com/kontakt/Octoprint-Filament-Reloaded/pull/29