kontakt/Octoprint-Filament-Reloaded #27: Can not run the plugins

ckd77, 2018-03-21:
Hello,

I installed under octopirnt 1.3.6 and a PI3 a day plugins. Configured to use PIN 7.

```
2018-03-21 19:05:54,253 - octoprint.startup - INFO - ******************************************************************************
2018-03-21 19:05:54,258 - octoprint.startup - INFO - Starting OctoPrint 1.3.6
2018-03-21 19:05:54,259 - octoprint.startup - INFO - ******************************************************************************
2018-03-21 19:05:54,325 - octoprint.startup - INFO - Blacklist processing done
2018-03-21 19:05:54,413 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2018-03-21 19:05:57,433 - octoprint.plugin.core - INFO - Found 17 plugin(s) providing 17 mixin implementations, 14 hook handlers
2018-03-21 19:05:57,603 - octoprint.server - INFO - Intermediary server started
2018-03-21 19:05:57,603 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2018-03-21 19:05:57,905 - octoprint.plugin.core - INFO - Found 17 plugin(s) providing 17 mixin implementations, 14 hook handlers
2018-03-21 19:05:57,939 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/pi/.octoprint/uploads...
2018-03-21 19:05:58,590 - octoprint.filemanager.storage - INFO - ... file metadata for /home/pi/.octoprint/uploads initialized successfully.
2018-03-21 19:05:58,621 - octoprint.plugins.filamentreload - INFO - Running RPi.GPIO version '0.6.3'
2018-03-21 19:05:58,783 - octoprint.plugins.softwareupdate - INFO - Loaded version cache from disk
2018-03-21 19:06:02,080 - octoprint.util.pip - INFO - Using "/home/pi/oprint/bin/python -m pip" as command to invoke pip
2018-03-21 19:06:04,586 - octoprint.util.pip - INFO - Version of pip is 9.0.1
2018-03-21 19:06:04,596 - octoprint.plugin.core - INFO - Initialized 17 plugin implementation(s)
2018-03-21 19:06:04,609 - octoprint.plugin.core - INFO - 17 plugin(s) registered with the system:
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins/announcements
|  Core Wizard (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins/corewizard
|  Cost Estimation (2.1.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_costestimation
|  CR-10 Leveling Plugin (2.0.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_CR10_Leveling
|  CuraEngine (<= 15.04) (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins/cura
|  Detailed Progress Plugin (0.1.4) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_detailedprogress
|  Discovery (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins/discovery
|  Fan Speed Control (0.1.7) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_fanspeedslider
|  Filament Sensor Reloaded (1.0.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filamentreload
|  Fullscreen Plugin (0.0.4) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_fullscreen
|  GcodeEditor (0.2.6) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_GcodeEditor
|  OctoPi Support Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins/octopi_support
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins/pluginmanager
|  Printer Stats (1.0.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_stats
|  Software Update (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins/softwareupdate
|  Tempsgraph Plugin (0.3.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_tempsgraph
|  Virtual Printer (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugins/virtual_printer
2018-03-21 19:06:06,371 - octoprint.environment - INFO - Detected environment is Python 2.7.9 under Linux (linux2). Details:
|  hardware:
|    cores: 4
|    freq: 1200.0
|    ram: 918216704
|  os:
|    id: linux
|    platform: linux2
|  plugins:
|    octopi_support:
|      model: 3B
|      revision: a02082
|      version: 0.14.0
|  python:
|    pip: 9.0.1
|    version: 2.7.9
|    virtualenv: /home/pi/oprint
2018-03-21 19:06:06,381 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/webassets...
2018-03-21 19:06:06,406 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/.webassets-cache...
2018-03-21 19:06:06,646 - octoprint.server - INFO - Shutting down intermediary server...
2018-03-21 19:06:06,818 - octoprint.server - INFO - Intermediary server shut down
2018-03-21 19:06:06,820 - octoprint.events - INFO - Processing startup event, this is our first event
2018-03-21 19:06:06,820 - octoprint.events - INFO - Adding 0 events to queue that were held back before startup event
2018-03-21 19:06:06,822 - octoprint.filemanager - INFO - Adding backlog items from all storage types to analysis queue...
2018-03-21 19:06:06,831 - octoprint.filemanager - INFO - Added 0 items from storage type "local" to analysis queue
2018-03-21 19:06:06,846 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance on octopi-cr10 for _http._tcp
2018-03-21 19:06:06,859 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance on octopi-cr10 for _octoprint._tcp
2018-03-21 19:06:06,862 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance on octopi-cr10 for SSDP
2018-03-21 19:06:06,863 - octoprint.server - INFO - Listening on http://127.0.0.1:5000
2018-03-21 19:06:06,892 - octoprint.plugins.filamentreload - INFO - Filament Sensor Reloaded started
2018-03-21 19:06:06,915 - octoprint.plugins.filamentreload - INFO - Setting up sensor.
2018-03-21 19:06:06,916 - octoprint.plugins.filamentreload - INFO - Using Board Mode
2018-03-21 19:06:06,918 - octoprint.plugins.filamentreload - INFO - Filament Sensor active on GPIO Pin [7]
2018-03-21 19:06:06,920 - octoprint.plugin - ERROR - Error while calling plugin filamentreload
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugin/__init__.py", line 225, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filamentreload/__init__.py", line 61, in on_after_startup
    self._setup_sensor()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filamentreload/__init__.py", line 55, in _setup_sensor
    GPIO.setup(self.pin, GPIO.IN, pull_up_down=GPIO.PUD_UP)
RuntimeError: No access to /dev/mem.  Try running as root!
```

I test in python directly, everything is ok.

```
pi@octopi-cr10:~/.octoprint $ python
Python 2.7.9 (default, Sep 17 2016, 20:26:04)
[GCC 4.9.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import RPi.GPIO as GPIO
>>> GPIO.setmode(GPIO.BOARD)
>>> configuration = GPIO.getmode()
>>> GPIO.setup(7, GPIO.IN)
>>> GPIO.setup(7, GPIO.OUT)
>>> GPIO.input(7)
1
>>> GPIO.input(7)
0
>>> GPIO.input(7)
1
>>>exit()
```

I also get this sometimes
```
2018-03-21 19:08:09,273 - octoprint.plugin - ERROR - Error while calling plugin filamentreload
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugin/__init__.py", line 225, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filamentreload/__init__.py", line 89, in on_event
    if event is Events.PRINT_STARTED and self.no_filament():
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filamentreload/__init__.py", line 81, in no_filament
    return GPIO.input(self.pin) != self.switch
RuntimeError: You must setup() the GPIO channel first
```

```
pi@octopi-cr10:~/.octoprint $ ls -l /dev/gpiomem
crw-rw-rw- 1 root root 248, 0 mars  21 19:05 /dev/gpiomem
```

I do not understand why live via python everything is oik, and directly via octoprint no.

EddyMI3d, 2018-03-21:
Which PI3? B or B+?
Gina already mentioned that B+ may not work correct because OctoPi is not configurated for the B+ yet.
(I just assume you installed OctoPi)

ckd77, 2018-03-27:
PI3 B

ckd77, 2018-03-31:
I just tested on a pi 2b and I have the same problem, same version of octoprint

ckd77, 2018-04-10:
After taking a new card sd descended octopi above sanbs do a rpi-update, so no update of the kernel plugins works wonderfully the first time. PI3 as PI2. The problem came from a version of the kernel as I thought at first.