kontakt/Octoprint-Filament-Reloaded #41: octoprint.plugins.filamentreload

arielramraz1, 2019-01-30:
Hello,

I installed the octoprint.plugins.filamentreload but its not working, in the logs I get:
2019-01-30 16:46:41,400 - octoprint.plugins.filamentreload - INFO - Sensor callback but no trigger state change.
2019-01-30 16:46:41,651 - octoprint.plugins.filamentreload - INFO - Sensor callback but no trigger state change.
2019-01-30 16:46:41,929 - octoprint.plugins.filamentreload - INFO - Sensor callback but no trigger state change.
2019-01-30 16:46:42,180 - octoprint.plugins.filamentreload - INFO - Sensor callback but no trigger state change.
2019-01-30 16:46:42,434 - octoprint.plugins.filamentreload - INFO - Sensor callback but no trigger state change.
2019-01-30 16:46:42,693 - octoprint.plugins.filamentreload - INFO - Sensor callback but no trigger state change.
2019-01-30 16:46:42,966 - octoprint.plugins.filamentreload - INFO - Sensor callback but no trigger state change.
2019-01-30 16:46:43,223 - octoprint.plugins.filamentreload - INFO - Sensor callback but no trigger state change.
2019-01-

Please assist.
Thanks

TofuNinja808, 2019-02-04:
Hello, I am having the same problem.  Did you get this resolved? 

2019-02-04 21:08:52,749 - octoprint.plugins.filamentreload - INFO - Filament Sensor active on GPIO Pin [24]
2019-02-04 21:09:42,682 - octoprint.plugins.filamentreload - INFO - Sensor callback but no trigger state change.
2019-02-04 21:09:47,712 - octoprint.plugins.filamentreload - INFO - Sensor callback but no trigger state change.
2019-02-04 21:09:49,985 - octoprint.plugins.filamentreload - INFO - Sensor callback but no trigger state change.
2019-02-04 21:09:50,583 - octoprint.plugins.filamentreload - INFO - Sensor callback but no trigger state change.
2019-02-04 21:09:51,985 - octoprint.plugins.filamentreload - INFO - Sensor callback but no trigger state change.
2019-02-04 21:09:55,122 - octoprint.plugins.filamentreload - INFO - Sensor callback but no trigger state change.
2019-02-04 21:09:56,225 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2019-02-04 21:11:21,359 - octoprint.plugins.filamentreload - INFO - Sensor callback but no trigger state change.

arielramraz1, 2019-02-05:
Not yet

Ariel

TimColby, 2019-02-08:
I have the same issue.  Confirmed that Pi can correctly detect the sensor state from GPIO.  The callback log appears repeatedly only when the switch is closed (filament is present).  If the switch is open, the log does not appear.  Octoprint does not react to sensor state.

jjvelar, 2019-02-10:
I have the same issue

MiTsCh13, 2019-02-10:
Same here..
Seems to me like everyone has got the same problem but the developers don't care about the plugin anymore.. :/

jvaldes, 2019-02-21:
The plugin worked correctly the first time I used the printer. From then on, the same error started to appear:
octoprint.plugins.filamentreload - INFO - Sensor callback but no trigger state change.

jvaldes, 2019-02-23:
After seeing the source code, I think there is nothing wrong with it. In my opinion it is a problem of hardware instability, probably because the cable from the sensor to the raspberry is quite long (in my case at least) and needs to stick to the power supply.
I have tried to coat the cable with aluminum foil and to install a ferrite toroid and the problem seems to have disappeared.
It is working correctly one day.

kashban, 2019-03-27:
Trying with aluminium foil... Don't have a ferrite toroid at hand.