kontakt/Octoprint-Filament-Reloaded #43: Filament runout not working

dompe-9, 2019-02-25:
I have an issue with this plugin. The filament sensor is recognize (/plugin/filamentreload/status OK), the print doesn't start when filament is out but during a print if I take off the filament nothing happen.
I'm on MKS GEN L with Marlin 1.1.9, RPI 3 B+ (Octoprint 1.3.10) filament sensor connect on the GPIO 17 (BCM).

dompe-9, 2019-03-05:
Anyone have a solution ?

codegrue, 2019-03-08:
Confirm I have the same behavior. State is read correctly (/plugin/filamentreload/status) but the printer never pauses.

GoJoeMoeJoe, 2019-03-23:
I'm in the same boat.

codegrue, 2019-03-24:
I switched to the NG version and it works perfectly.

GoJoeMoeJoe, 2019-03-25:
That's funny... That one didn't work for me and I ended up here. ;) I'll give it another go.

codegrue, 2019-03-25:
Yea I wish these three would get together and just publish one that works better. The NG one doesn't seem to have the REST api to test it out, but aside from that it's pausing correctly.

kashban, 2019-03-27:
No luck with either one. GPIO Pin 24 + ground. Sensor is recognized and prevents starting the print if no filament is there, but fails to do so mid print.

codegrue, 2019-03-27:
My sensor has a 5v pin also since it has a light on it. Otherwise, I was getting the same behavior that it wouldn't start, but also wouldn't pause.

kashban, 2019-03-27:
I use a simple microswitch. I get a reaction but the print does not pause,

GoJoeMoeJoe, 2019-03-27:
> I use a simple Micro Switch ~the print does not pause,

That is what I'm doing also. Using https://pinout.xyz/pinout/pin26_gpio7 and ground. Probably need to look at the logs. :/

kashban, 2019-03-27:
It was an  hardware issue on my part, bad soldering the switch. I have it working with FilamentReloadedNG now and I presume it works with this plugin too. NG has a debug mode, perhaps that could help find the issue.

GoJoeMoeJoe, 2019-03-28:
> I presume it works with this plugin too.

Could you try this plugin?
Also what pin are you using?

kashban, 2019-03-28:
Sorry, I just started a 16 hour print. Will do tomorrow when it's finished.

I use GPIO 24 and the ground beside it.

RomRider, 2019-11-20:
This is fixed in a fork of this plugin I developed and which also supports jam sensor if required.
You can find it here: https://github.com/RomRider/Octoprint-Filament-Revolutions

Just import https://github.com/RomRider/Octoprint-Filament-Revolutions/archive/master.zip in Octoprint. I'll add it the the official plugin repo soon.

eightheads, 2019-11-25:
> This is fixed in a fork of this plugin I developed and which also supports jam sensor if required.
> You can find it here: https://github.com/RomRider/Octoprint-Filament-Revolutions
>
> Just import [...]

@RomRider could the Bigtreetech Smart Filament Sensor be used with your fork? I don't see any information how to hookup and sense a jam with your setup.
https://www.amazon.com/BIGTREETECH-Filament-Sensor-Detection-Printer/dp/B07Z7Y5VY9/

RomRider, 2019-11-25:
As far as I know this module only works by connecting it to the board of your printer

In my case, the jam sensor has to act like a switch (open or closed when a jam occurs). If this module can do that, then it would be compatible.

eightheads, 2019-11-25:
> As far as I know this module only works by connecting it to the board of your printer
> 
> In my case, the jam sensor has to act like a switch (open or closed when a jam occurs). If this module can do that, then it would be compatible.

Do you have an example of what sensor you are using with your fork? Feel free to email me eight.heads@gmail.com

RomRider, 2019-11-25:
Just a basic switch: https://www.amazon.fr/gp/product/B07912Q45W/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1

eightheads, 2019-11-25:
> Just a basic switch: https://www.amazon.fr/gp/product/B07912Q45W/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1

How does that sense a jam then?