kontakt/Octoprint-Filament-Reloaded #44: normally closed support

frantisekbrabec, 2019-03-05:
My specific sensor (seems to be the same used by TH3D - https://www.th3dstudio.com/product/ezout-v2-filament-sensor-only-replacement-spare/) goes HIGH when filament is present and LOW when absent.

I wired it the most straightforward way (Vcc-Vcc, GND-GND, signal-GPIO; no resistors, etc).

The only way I can read correct GPIO values is when I set up the IO with pull_up_down=GPIO.PUD_DOWN option. However, the plugin code has pull_up_down=GPIO.PUD_UP hardwired.  It probably should be parameter driven, perhaps derived from the existing 'switch' parameter?

guysoft, 2019-09-24:
It looks like this exists now. should this issue be closed?

frantisekbrabec, 2019-09-24:
Not really, it still has PUD_UP hardwired:

GPIO.setup(self.pin, GPIO.IN, pull_up_down=GPIO.PUD_UP)

do you see something else somewhere?