kontakt/Octoprint-Filament-Reloaded #55: Is this plugin abandoned?

mrseeker, 2020-05-14:
Since I see that there are no replies on any issue since Sep 25, 2019 and the latest commit was done on Jan 4, 2019, is this plugin still being maintained by anyone?

linux-paul, 2020-05-31:
I fixed some bugs myself, but I didn't get a segmentation fault or how a python script is able to produce it?

mrseeker, 2020-05-31:
@linux-paul the python.GPIO implementation triggers a segfault when it closes the connection. I could trigger it after cancelling a print. Why it happens, I am unsure, but it might be fixed in a future implementation of gpio.

linux-paul, 2020-05-31:
Which connection has been closed? Which log shows the fault? In a first attempt I would say the fault is created by the RPI.GPIO itself. The plugin just creates a GPIO Object and connect / disconnect the event handler.

mrseeker, 2020-05-31:
When gpio.close() is called, a segfault gets triggered and the python application (octoprint) gets terminated. This is not seen in any log except stderr. To reproduce it, you need to start octoprint manually (not as a service). Although gpio might be the root cause of the issue, this plugin relies on gpio and is therefore broken. My original question was if there was a workaround, but since nobody has been working on it, nobody bothered to fix it, it stays broken until the guys behind gpio fix it or until there is a workaround.